nix-community/robotnix #214: Could we have a more generic `etc`  modules to put files in any path?

Vonfry, 2023-09-17:
The `etc` module is implemented though `robotnix/etcfiles` sources. For example, some apps use config files under `/sdcard/` directly which is other than app resouces and etc.

Atemu, 2023-09-17:
That won't work as /sdcard is on the userdata partition. etcfiles are put into the system image's /etc.

Everything robotnix does concerns system partition currently.

closed 2023-09-17 by Atemu

Atemu, 2023-09-17:
Declarative config of userdata things is something I'd *love* to have aswell but that'd be its own separate project; similar to NixOS and home-manager.

The most robotnix could do is bake running that other project into the system image so that you wouldn't have to install it imperatively.