nix-community/robotnix #218: Update Nixpkgs to something less ancient

Atemu, 2023-11-09:
Currently, robotnix still uses 21.11:

https://github.com/nix-community/robotnix/blob/73d6b0b142e3f2844e6caaf820e8be3bd8cde2c6/flake.nix#L5

We should update that to at least 22.11 which is the last release with python2 which is still used in quite a few places here.

ajs124, 2023-11-09:
> which is the last release with python2

`python2` still exists on nixpkgs master and has been updated quite recently https://github.com/NixOS/nixpkgs/pull/256132.

It's just been marked as insecure, because it's EOL upstream for some years now, but it's still getting (3rd party) updates, so IMHO this shouldn't hold us back from using a nixpkgs newer than 22.11.

Atemu, 2023-11-09:
My WIP for reference: https://github.com/nix-community/robotnix/pull/220

Atemu, 2024-10-26:
Fixed by https://github.com/nix-community/robotnix/pull/231

closed 2024-10-26 by Atemu