nix-community/robotnix #223: How to deal with a version-bootloader mismatch?

doronbehar, 2024-01-22:
I'm getting the following error when I run `fastboot update`:

```
$ fastboot -w --skip-reboot update sargo-img-2022022022.zip
--------------------------------------------
Bootloader Version...: b4s4-0.4-8048689
Baseband Version.....: g670-00145-220106-B-8048689
Serial Number........: 03DAY1V2U5
--------------------------------------------
extracting android-info.txt (0 MB) to RAM...
Checking 'product'                                 OKAY [  0.058s]
Checking 'version-bootloader'                      FAILED

Device version-bootloader is 'b4s4-0.4-8048689'.
Update requires 'b4s4-0.4-7758097'.

fastboot: error: requirements not met!
```

What can I do?

Atemu, 2024-01-22:
No idea, sorry. I have not personally dealt with downgrade protection of Pixel phones.

Perhaps it could be worth trying to get a newer version built.

doronbehar, 2024-01-22:
Could you point me to where the version of the bootloader defined in robotnix? Meaning, to the point where the bootloader's src is built from? `git grep bootloader` gave me only some relevant lines in `components.json` so it seems...

doronbehar, 2024-01-23:
Hmm so maybe I can provide robotnix a different AOSP source? With only the bootloader version updated?

Atemu, 2024-01-23:
Sure. Find the relative path and override it in `source.dirs`.

doronbehar, 2024-01-28:
I forgot to ran `./flash-base.sh`... :facepalm: Apparently though, I still can't get the `fastboot update` command to work:

```
Warning: skip copying bootloader_b image avb footer (bootloader_b partition size: 0, bootloader_b image size: 8578728).
Sending 'bootloader_b' (8377 KB)                   OKAY [  0.409s]
Writing 'bootloader_b'                             (bootloader) Flashing Pack version b4s4-0.4-7758097
(bootloader) Flashing partition table for Lun = 0
(bootloader) Flashing partition xbl_b
(bootloader) Flashing partition xbl_config_b
(bootloader) Flashing partition tz_b
(bootloader) Flashing partition aop_b
(bootloader) Flashing partition hyp_b
(bootloader) Flashing partition keymaster_b
(bootloader) Flashing partition cmnlib_b
(bootloader) Flashing partition cmnlib64_b
(bootloader) Flashing partition abl_b
(bootloader) Flashing partition devcfg_b
(bootloader) Flashing partition qupfw_b
(bootloader) Flashing partition storsec_b
OKAY [  0.306s]
Finished. Total time: 1.097s
Rebooting into bootloader                          OKAY [  0.045s]
Finished. Total time: 0.095s
Warning: skip copying radio_b image avb footer (radio_b partition size: 0, radio_b image size: 75116684).
Sending 'radio_b' (73356 KB)                       OKAY [  2.790s]
Writing 'radio_b'                                  (bootloader) Flashing Pack version SSD:g670-00134-210928-B-7773694
(bootloader) Flashing partition modem_b
OKAY [  0.479s]
Finished. Total time: 3.644s
Rebooting into bootloader                          OKAY [  0.041s]
Finished. Total time: 0.092s
--------------------------------------------
Bootloader Version...: b4s4-0.4-7758097
Baseband Version.....: g670-00134-210928-B-7773694
Serial Number........: 03DAY1V2U5
--------------------------------------------
extracting android-info.txt (0 MB) to RAM...
Checking 'product'                                 OKAY [  0.059s]
Checking 'version-bootloader'                      OKAY [  0.058s]
Checking 'version-baseband'                        OKAY [  0.058s]
Setting current slot to 'b'                        OKAY [  0.327s]
extracting boot.img (64 MB) to disk... took 0.179s
archive does not contain 'init_boot.img'
extracting dtbo.img (8 MB) to disk... took 0.013s
archive does not contain 'dt.img'
archive does not contain 'pvmfw.img'
archive does not contain 'recovery.img'
extracting vbmeta.img (0 MB) to disk... took 0.000s
archive does not contain 'vbmeta_system.img'
archive does not contain 'vbmeta_vendor.img'
archive does not contain 'vendor_boot.img'
archive does not contain 'vendor_kernel_boot.img'
extracting super_empty.img (0 MB) to disk... took 0.000s
archive does not contain 'boot_other.img'
archive does not contain 'odm.img'
archive does not contain 'odm_dlkm.img'
extracting product.img (345 MB) to disk... took 1.276s
extracting system.img (775 MB) to disk... took 2.503s
archive does not contain 'system_dlkm.img'
extracting system_ext.img (173 MB) to disk... took 0.547s
extracting system_other.img (71 MB) to disk... took 0.184s
extracting vendor.img (485 MB) to disk... took 1.500s
archive does not contain 'vendor_dlkm.img'
archive does not contain 'vendor_other.img'
extracting boot.img (64 MB) to disk... took 0.184s
archive does not contain 'boot.sig'
Sending 'boot_b' (65536 KB)                        OKAY [  2.480s]
Writing 'boot_b'                                   OKAY [  0.357s]
extracting dtbo.img (8 MB) to disk... took 0.032s
archive does not contain 'dtbo.sig'
[... 18 lines cut ...]
Sending sparse 'product_b' 1/2 (262140 KB)         OKAY [  9.719s]
Writing 'product_b'                                OKAY [  1.446s]
Sending sparse 'product_b' 2/2 (91584 KB)          OKAY [  3.394s]
Writing 'product_b'                                OKAY [  0.568s]
extracting system.img (775 MB) to disk... took 2.514s
archive does not contain 'system.sig'
Resizing 'system_b'                                OKAY [  0.006s]
Sending sparse 'system_b' 1/4 (262140 KB)          OKAY [  9.701s]
Writing 'system_b'                                 OKAY [  1.432s]
Sending sparse 'system_b' 2/4 (262140 KB)          OKAY [  9.824s]
Writing 'system_b'                                 OKAY [  1.406s]
Sending sparse 'system_b' 3/4 (257820 KB)          OKAY [  7.909s]
Writing 'system_b'                                 OKAY [  1.341s]
Sending sparse 'system_b' 4/4 (12448 KB)           OKAY [  0.407s]
Writing 'system_b'                                 OKAY [  0.096s]
extracting system_ext.img (173 MB) to disk... took 0.563s
archive does not contain 'system_ext.sig'
Resizing 'system_ext_b'                            OKAY [  0.007s]
Sending 'system_ext_b' (177196 KB)                 OKAY [  5.677s]
Writing 'system_ext_b'                             OKAY [  0.978s]
extracting system_other.img (71 MB) to disk... took 0.188s
archive does not contain 'system_other.sig'
Sending 'system_a' (3190784 KB)                    FAILED (remote: 'Invalid size')
fastboot: error: Command failed
```

doronbehar, 2024-04-28:
Reporting here my progress:

I found [this chinese forum thread](https://www-cnblogs-com.translate.goog/rab3it/p/17844308.html?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en) indicating that `fastboot` from the android tools package was buggy, and downgrading it should help an error similar to the one reported above.

I discovered the error they report there by trying to flash [Google's image for the pixel 3a](https://dl.google.com/dl/android/aosp/sargo-sp2a.220505.008-factory-071e368a.zip). I tried [the latest android-tools 35.0.1 release](https://github.com/nmeum/android-tools/releases/tag/35.0.1), and I managed to flash Google's image, and it booted successfully. However, after I ran `./flash-all.sh` from the zip directory produced by robotnix, it finished successfully, but produced a device that tries to boot twice, fails, and eventually ends in a bootloader mode :(.