nix-community/robotnix #224: "stack overflow; max-call-depth exceeded" when building generateKeysScript

CyberShadow, 2024-03-09:
```console
$ cat ../in2023.nix      
{
  device = "instantnoodlep";
  flavor = "lineageos";
}

$ nix-build --arg configuration ../in2023.nix -A generateKeysScript -o generate-keys --show-trace
error:
       … while calling the 'derivationStrict' builtin
         at <nix/derivation-internal.nix>:9:12:
            8|
            9|   strict = derivationStrict drvAttrs;
             |            ^
           10|

       … while evaluating derivation 'generate_keys.sh'
         whose name attribute is located at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/pkgs/stdenv/generic/make-derivation.nix:205:7

       … while evaluating attribute 'text' of derivation 'generate_keys.sh'
         at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/pkgs/build-support/trivial-builders.nix:117:16:
          116|     runCommand name
          117|       { inherit text executable checkPhase meta;
             |                ^
          118|         passAsFile = [ "text" ];

       … while evaluating derivation 'android-key-tools'
         whose name attribute is located at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/pkgs/stdenv/generic/make-derivation.nix:205:7

       … while evaluating attribute 'buildCommand' of derivation 'android-key-tools'
         at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/pkgs/build-support/trivial-builders.nix:74:14:
           73|       name = lib.strings.sanitizeDerivationName name;
           74|       inherit buildCommand;
             |              ^
           75|       passAsFile = [ "buildCommand" ]

       … from call site
         at /home/vladimir/work/android-build/robotnix/modules/signing.nix:205:14:
          204|
          205|         cp ${config.source.dirs."development".src}/tools/make_key $out/bin/make_key
             |              ^
          206|         substituteInPlace $out/bin/make_key --replace openssl ${lib.getBin pkgs.openssl}/bin/openssl

       … while calling 'g'
         at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/lib/attrsets.nix:301:19:
          300|           g =
          301|             name: value:
             |                   ^
          302|             if isAttrs value && cond value

       … from call site
         at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/lib/attrsets.nix:304:20:
          303|               then recurse (path ++ [name]) value
          304|               else f (path ++ [name]) value;
             |                    ^
          305|         in mapAttrs g set;

       … while calling anonymous lambda
         at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/lib/modules.nix:171:72:
          170|           # For definitions that have an associated option
          171|           declaredConfig = mapAttrsRecursiveCond (v: ! isOption v) (_: v: v.value) options;
             |                                                                        ^
          172|

       … while evaluating the option `source.dirs.development.src':

       … from call site
         at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/lib/modules.nix:578:35:
          577|       # yield a value computed from the definitions
          578|       value = if opt ? apply then opt.apply res.mergedValue else res.mergedValue;
[... 166 lines cut ...]
          382|   /* Like `zipAttrsWith' with `(name: values: values)' as the function.

       … while calling 'zipAttrsWithNames'
         at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/lib/attrsets.nix:366:33:
          365|   */
          366|   zipAttrsWithNames = names: f: sets:
             |                                 ^
          367|     listToAttrs (map (name: {

       … from call site
         at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/lib/lists.nix:58:34:
           57|         then nul
           58|         else op (elemAt list n) (fold' (n + 1));
             |                                  ^
           59|     in fold' 0;

       (11079 duplicate frames omitted)

       error: stack overflow; max-call-depth exceeded
       at /nix/store/d0dl9ch4hhxkqikc1dw3868j95gsbb0h-source/lib/attrsets.nix:381:46:
          380|   */
          381|   zipAttrsWith = f: sets: zipAttrsWithNames (concatMap attrNames sets) f sets;
             |                                              ^
          382|   /* Like `zipAttrsWith' with `(name: values: values)' as the function.
```

CyberShadow, 2024-03-09:
```
2dcae7d82f5ef0b373413d2fbfc1001141561c74 is the first good commit
commit 2dcae7d82f5ef0b373413d2fbfc1001141561c74
Author: pennae <github@quasiparticle.net>
Date:   Sat Dec 25 15:20:26 2021 +0100

    lib/attrsets: use builtins.zipAttrsWith if available

 lib/attrsets.nix | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)
bisect found first good commit
```

https://github.com/NixOS/nixpkgs/commit/2dcae7d82f5ef0b373413d2fbfc1001141561c74
https://github.com/NixOS/nixpkgs/pull/152046

Looks like it wasn't even intended as a bugfix.

Atemu, 2024-03-09:
If we can get that exact version to build, feel free to update the lock.

See https://github.com/nix-community/robotnix/issues/218

c2vi, 2024-04-23:
just increasing the max-call-depth by adding `--option max-call-depth 1000000` to the `nix build` command solved it for me as a workaround