nix-community/robotnix #267: Declarative ADB keys; adb at boot

Atemu, 2024-12-23:
For debugging purposes, it'd be incredibly useful to be able to declaratively enable ADB on boot and have your keys be trusted.

The following props need to be set in order to enable ADB on boot:

```
persist.service.adb.enable=1
persist.service.debuggable=1
persist.sys.usb.config=mtp,adb
```

There probably is a way to set these but I haven't found it yet or how to do it via robotnix.

Trusted ADB keys go in `/data/misc/adb/adb_keys`. This is a lot harder to bake into the ROM as that's a runtime state directory. It also gets reset frequently. Perhaps there's a way to bake in the key into the ROM though.

samueldr, 2024-12-24:
Not sure if any helpful, but IIRC setting the *build variant* to `eng` makes `adb` available at boot. (Or was it `userdebug`?)

 - https://source.android.com/docs/setup/build/building#choose-a-target
 - https://source.android.com/docs/setup/create/new-device#build-layers

I'm thinking a bit more here, and I seem to recall that maybe LinageOS made their `userdebug` builds differ in semantics from AOSP... aaaaaaaaand:

 - https://review.lineageos.org/c/LineageOS/android_build/+/317095

Not sure if any of those hints will help. Those, in any way, don't help with the adb key.

Atemu, 2024-12-24:
This makes sense because most LOS devices are userdebug rather than user.

I'm wondering though whether it's possible to override this setting somehow. I guess you can always patch the sources...

samueldr, 2024-12-24:
:thinking: I don't grok the big picture enough to know how `post_process_props` ends-up used.

 - https://github.com/search?q=org%3ALineageOS%20%2Fpost_process_props%2F&type=code

I don't see any usage that would make any keys disallowed

 - https://github.com/LineageOS/android_build/blob/9531801b02c499820c129fb0ef501b1c76c34550/tools/post_process_props.py#L241
 - https://github.com/search?q=org%3ALineageOS%20%2Fdisallowed.keys%2F&type=code

So that leaves the `mangle_build_prop` function that could throw a wrench into any plans...

 - https://github.com/LineageOS/android_build/blob/9531801b02c499820c129fb0ef501b1c76c34550/tools/post_process_props.py#L30-L47

Except that it looks like it should be fine:

Only if `ro.adb.secure` does it setup `persist.sys.usb.config`. So at the very least it shouldn't be fighting against additional shenanigans around `build.prop`.

So, my gut feeling here is that it's “just” (lol) a matter of having a way to override `build.prop` props.

samueldr, 2024-12-24:
Speaking of `build.prop` files... I was wondering where it would be... and I don't like what I'm seeing:

 - https://github.com/LineageOS/android_build/blob/9531801b02c499820c129fb0ef501b1c76c34550/core/sysprop.mk#L29-L32

~~Embedding bash snippets into a higher level build system? Who would do that?~~

* * *

Ignoring the satire... `ADDITIONAL_[...]_PROPERTIES` are likely what is desired.

 - https://github.com/search?q=org%3ALineageOS+%2FADDITIONAL_SYSTEM_PROPERTIES%2F&type=code
 - https://github.com/search?q=org%3ALineageOS+%2FADDITIONAL_%5B%5E_%5D%2B_PROPERTIES%2F&type=code
 - https://github.com/LineageOS/android_build/commit/80f164c0b3933fc13d7de87c44fd83af59d0a7c5
 - https://github.com/LineageOS/android_build/blob/9531801b02c499820c129fb0ef501b1c76c34550/buildspec.mk.default#L88-L91

Atemu, 2024-12-25:
Thanks a bunch for the research notes!

How would I set those additional properties in Robotnix though?

samueldr, 2024-12-25:
I don't *know*, but I believe this is the current best(?) mechanism for LineageOS:

 - https://review.lineageos.org/c/LineageOS/android_build/+/285808/1
 - https://github.com/LineageOS/android_build/commit/24304ab838226ef2546aee95f29a8ead5004ee03
 - https://github.com/search?q=path%3ABoardConfigExtra.mk&type=code

So something like `source.dirs."vendor/extra"`, which can be built via a Nix expression.

Though I wonder if AOSP has such a thing. If it doesn't, it would require patching the AOSP `android_build` to do so, but that's not a lot to implement tbf, so it should be trivial to add there too.